araffin/learning-to-drive-in-5-minutes #2: Input to the Donkey Simulator

kk2491, 2019-02-06:
Hi,

Thanks for the repository.

I have downloaded the code and started running the simulator. Could you please help me in understanding the below queries.

1. What are the inputs to the algorithm?
2. If camera is the input, how can I extract the camera images?

Thank you,
KK

araffin, 2019-02-06:
>What are the inputs to the algorithm?

Input to the VAE is an image, input to the RL algo is a vector (encoded image using the VAE).

>If camera is the input, how can I extract the camera images?

I don't really understand that question... Simulator directly sends the image.

kk2491, 2019-02-06:
@araffin Thanks for the reply. 
I wanted to capture the images during the training process.

araffin, 2019-02-06:
There is alerady the [recorder wrapper](https://github.com/araffin/learning-to-drive-in-5-minutes/blob/master/teleop/recorder.py) for that.

kk2491, 2019-02-07:
@araffin Yep.. Got you.. Thank you so much for the swift response. I have been working on this project will post here if any queries.

closed 2019-06-09 by araffin