araffin/learning-to-drive-in-5-minutes #3: AttributeError: module 'gym.spaces' has no attribute 'prng'

lamhoangtung, 2019-02-09:
Hi @araffin. Thanks for the great repo. I'm trying to train everything from scratch based on your work.

I collected images using the teleoperation mode and starting to train the VAE but the following error come up when I tried to run `python3 -m vae.train --n-epochs 50 --verbose 0 --z-size 64 -f ./data/vae/`

```
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/linus/learning-to-drive-in-5-minutes/vae/train.py", line 31, in <module>
    set_global_seeds(args.seed)
  File "/home/linus/.local/lib/python3.6/site-packages/stable_baselines/common/misc_util.py", line 92, in set_global_seeds
    gym.spaces.prng.seed(seed)
AttributeError: module 'gym.spaces' has no attribute 'prng'
```

Everything is up-to-date with the latest code on `master`. I installed all the dependencies with `pip3 install -t requirement`. Also, my system don't have any virtual environment or anaconda. I tried to reinstall `gym` a couple time via pip but it still not working.

Any ideals how to solve this problem ? Thanks a lot.

araffin, 2019-02-09:
Hello,
this error is due to open ai breaking gym api.
Two solutions:
- downgrade your gym package to 0.10.6 
- install stable-baselines from source using the `enhancements` branch (this PR (https://github.com/hill-a/stable-baselines/pull/185) fixes the issue  and will be merged soon with master branch) if you wait a bit, you will be able to use master branch of stable-baselines.

lamhoangtung, 2019-02-10:
Thanks @araffin for the quick reply. I installed `stable-baselines` from `master` branch and everything is working great now. I put this command here just in case anyone needed it ;)
```
pip3 install git+https://github.com/hill-a/stable-baselines
```

closed 2019-02-10 by lamhoangtung

araffin, 2019-02-11:
Perfect.

Note: the command is already in the doc ;)