araffin/learning-to-drive-in-5-minutes #13: Try to learn from pixels

qipengliuNTU, 2019-07-15:
From my understanding, if I don't set the arguments 'vae_path' and 'random_features', the car will learn from pixels, as mentioned in 'train.py':
```
if args.vae_path != '':
    print("Loading VAE ...")
    vae = load_vae(args.vae_path)
elif args.random_features:
    print("Randomly initialized VAE")
    vae = load_vae(z_size=Z_SIZE)
    # Save network
    args.save_vae = True
else:
    print("Learning from pixels...")
```

However, when I run the following command, 
` python train.py --algo sac  -n 5000 --teleop `
some error message comes up:
```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/automan/learning-to-drive-in-5-minutes/teleop/teleop_client.py", line 274, in main_loop
    reward, done, info)
  File "/home/automan/learning-to-drive-in-5-minutes/donkey_gym/envs/vae_env.py", line 156, in postprocessing_step
    observation = np.concatenate((observation, self.command_history), axis=-1)
ValueError: all the input arrays must have same number of dimensions
```

Any advice on this problem? Thanks very much.

araffin, 2019-07-15:
Hello,
If you train from pixels, you should set `N_COMMAND_HISTORY` to zero.
See warning here: https://github.com/araffin/learning-to-drive-in-5-minutes/blob/master/donkey_gym/envs/vae_env.py#L92

qipengliuNTU, 2019-07-15:
Oh~ I didn't find this warning message. 
Now I can train the car from pixels.
Thanks very much for your help!!!
I am closing this issue.

closed 2019-07-15 by qipengliuNTU

ChunJyeBehBeh, 2020-03-02:
@qipengliuNTU Hello Did you manage to train the model with the image input?

I set the **N_COMMAND_HISTORY** to zero but I encountered the following issue.
`ValueError: could not broadcast input array from shape (38400) into shape (80,160,3)`

ChunJyeBehBeh, 2020-03-06:
@qipengliuNTU @araffin I tried to edit `postprocessing_step` and `reset` in vae_env.py so that it only return observation.flatten() when using VAE training. The code can run but seem have some issue on the training. The model seem didn't learn and keep go off the track when it start a new episode....