araffin/learning-to-drive-in-5-minutes #16: Cannot control car in Teleoperation mode

srs3, 2019-07-28:
Hi, 

I am attempting to train the VAE. However, I cannot control the car in teleoperation mode, it remains still at the start line. Examining the images captured, they are all the same. When I exit the simulation and force quit the application from running in my terminal, I get a  message that says "the current event queue and the main event queue are not the same".

Log:
~~~~
(base) Surajs-MacBook-Pro:DDPGDonkey suraj$ python -m teleop.teleop_client --record-folder /logs/ddpg 
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Starting DonkeyGym env
Donkey subprocess started
Binding to ('0.0.0.0', 9091)
Waiting for sim to start...if the simulation is running, press EXIT to go back to the menu
2019-07-28 11:08:19.114 donkey_sim[5468:423087] Could not find image named 'ScreenSelector'.
Waiting for sim to start...if the simulation is running, press EXIT to go back to the menu
Waiting for sim to start...if the simulation is running, press EXIT to go back to the menu
2019-07-28 11:08:25.212 donkey_sim[5468:423087] S22D300 preferred device: AMD Radeon Pro 560 (high power)
2019-07-28 11:08:25.212 donkey_sim[5468:423087] Metal devices available: 2
2019-07-28 11:08:25.213 donkey_sim[5468:423087] 0: AMD Radeon Pro 560 (high power)
2019-07-28 11:08:25.213 donkey_sim[5468:423087] 1: Intel(R) HD Graphics 630 (low power)
2019-07-28 11:08:25.213 donkey_sim[5468:423087] Using device AMD Radeon Pro 560 (high power)
Waiting for sim to start...if the simulation is running, press EXIT to go back to the menu
Got a new client ('127.0.0.1', 51347)
Scene Selection Ready
Connection dropped
Got a new client ('127.0.0.1', 51348)
Recorder current idx: 0
2019-07-28 11:08:31.487 python[5462:423543] WARNING: nextEventMatchingMask should only be called from the Main Thread! This will throw an exception in the future.
2019-07-28 11:08:31.531 python[5462:423543] WARNING: NSWindow drag regions should only be invalidated on the Main Thread! This will throw an exception in the future. Called from (
	0   AppKit                              0x00007fff2bfda607 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 378
	1   AppKit                              0x00007fff2bfd79f7 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1479
	2   AppKit                              0x00007fff2bfd742a -[NSWindow initWithContentRect:styleMask:backing:defer:] + 45
	3   libSDL-1.2.0.dylib                  0x000000011a277611 -[SDL_QuartzWindow initWithContentRect:styleMask:backing:defer:] + 276
	4   libSDL-1.2.0.dylib                  0x000000011a274fcd QZ_SetVideoMode + 1150
	5   libSDL-1.2.0.dylib                  0x000000011a26c50e SDL_SetVideoMode + 911
	6   display.cpython-37m-darwin.so       0x000000011a31099c pg_set_mode + 348
	7   python                              0x000000010ba3a1b8 _PyMethodDef_RawFastCallKeywords + 392
	8   python                              0x000000010bb76e42 call_function + 306
	9   python                              0x000000010bb74aec _PyEval_EvalFrameDefault + 46092
	10  python                              0x000000010ba398d5 function_code_fastcall + 117
	11  python                              0x000000010ba3cce2 method_call + 130
	12  python                              0x000000010ba3a752 PyObject_Call + 130
	13  python                              0x000000010bb74d58 _PyEval_EvalFrameDefault + 46712
	14  python                              0x000000010ba398d5 function_code_fastcall + 117
	15  python                              0x000000010bb76dc7 call_function + 183
	16  python                              0x000000010bb74a56 _PyEval_EvalFrameDefault + 45942
	17  python                              0x000000010ba398d5 function_code_fastcall + 117
	18  python                              0x000000010bb76dc7 call_function + 183
	19  python                              0x000000010bb74a56 _PyEval_EvalFrameDefault + 45942
	20  python                              0x000000010ba398d5 function_code_fastcall + 117
	21  python                              0x000000010ba3cce2 method_call + 130
	22  python                              0x000000010ba3a752 PyObject_Call + 130
	23  python                              0x000000010bc588cb t_bootstrap + 123
	24  python                              0x000000010bbdf707 pythread_wrapper + 39
	25  libsystem_pthread.dylib             0x00007fff5aaad2eb _pthread_body + 126
	26  libsystem_pthread.dylib             0x00007fff5aab0249 _pthread_start + 66
	27  libsystem_pthread.dylib             0x00007fff5aaac40d thread_start + 13
)
^CConnection dropped
^CTraceback (most recent call last):
  File "//anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "//anaconda3/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/DDPGDonkey/teleop/teleop_client.py", line 481, in <module>
    env.exit()
  File "/DDPGDonkey/teleop/teleop_client.py", line 191, in exit
    self.env.reset()
  File "DDPGDonkey/teleop/recorder.py", line 36, in reset
    obs = self.env.reset()
[... 29 lines cut ...]
	9   CoreFoundation                      0x00007fff2e963a01 __CFRunLoopRun + 2289
	10  CoreFoundation                      0x00007fff2e962ebe CFRunLoopRunSpecific + 455
	11  HIToolbox                           0x00007fff2dbc21ab RunCurrentEventLoopInMode + 292
	12  HIToolbox                           0x00007fff2dbc1ee5 ReceiveNextEventCommon + 603
	13  HIToolbox                           0x00007fff2dbc1c76 _BlockUntilNextEventMatchingListInModeWithFilter + 64
	14  AppKit                              0x00007fff2bf5a79d _DPSNextEvent + 1135
	15  AppKit                              0x00007fff2bf5948b -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1361
	16  libSDL-1.2.0.dylib                  0x000000011a272ee8 QZ_PumpEvents + 231
	17  libSDL-1.2.0.dylib                  0x000000011a251dbb SDL_PumpEvents + 35
	18  libSDL-1.2.0.dylib                  0x000000011a251fa6 SDL_EventState + 211
	19  libSDL-1.2.0.dylib                  0x000000011a271878 SDL_JoystickEventState + 41
	20  joystick.cpython-37m-darwin.so      0x000000011a4c604b joy_autoquit + 75
	21  base.cpython-37m-darwin.so          0x000000011a242ee4 _pg_quit + 308
	22  base.cpython-37m-darwin.so          0x000000011a242989 pg_quit + 9
	23  python                              0x000000010ba391bc _PyMethodDef_RawFastCallDict + 140
	24  python                              0x000000010ba3a8fd PyCFunction_Call + 61
	25  python                              0x000000010bc50a4d atexit_callfuncs + 141
	26  python                              0x000000010bbbeef3 Py_FinalizeEx + 67
	27  python                              0x000000010bbf8314 pymain_main + 7060
	28  python                              0x000000010ba0c66d main + 125
	29  libdyld.dylib                       0x00007fff5a8b93d5 start + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
~~~~

srs3, 2019-07-29:
It appears as though the issue is with pygame. Mac OS does not allow pygame to run on a separate thread, which is what the code in the teleop_client.py appears to do. 

I am a little stuck on how to resolve this, any suggestion would be massively appreciated. 

Thanks!

srs3, 2019-07-30:
My above suspicion was correct - temporarily resolved the issue by running the main_loop without assigning it to a thread.

closed 2019-07-30 by srs3

zhl001, 2019-09-27:
Sorry, I tried to run the main_loop without assigning it to a thread, but it didn't work.