araffin/learning-to-drive-in-5-minutes #19: [Question] What features are missing to make it possible to train the car in level 1 without using teleop?

FrancescoTerrosi, 2019-11-06:
As the title says:

I would like to expand this project in order to make it possible for the car to learn by itself in every environment, could you please tell me why this is not permitted right now? thanks

araffin, 2019-11-06:
Hello,

This comes from the simulator: it does not provide a distance to the center of the road, so you cannot automatically stop the episode and must use teleop.

See [here](https://github.com/araffin/learning-to-drive-in-5-minutes/blob/master/donkey_gym/envs/donkey_sim.py#L272) in the code (cross track error)

FrancescoTerrosi, 2019-11-06:
Yes I read that comment and it confuses me because in the repo's README you're saying that cte works only on level 0 but I understand that comment as if cte is available in all the 3 tracks. Also I'm not getting any error so is it actually implemented for now?
Last question: does the simulator provide the distance for track 0 and not for the others? 
Thank you :)

araffin, 2019-11-06:
>Also I'm not getting any error so is it actually implemented for now?
>Last question: does the simulator provide the distance for track 0 and not for the others?

It is not, it just returns zero all the time. But this is not related to this repo.

FrancescoTerrosi, 2019-11-06:
Okay thank you, I thought you were just missing some preprocessing or stuff like that.

Cheers

closed 2019-11-06 by FrancescoTerrosi