araffin/learning-to-drive-in-5-minutes #21: AssertionError on enjoy.py

iamycee, 2020-01-20:
AssertionError: No model found for sac on DonkeyVae-v0-level-0, path: logs/sac/DonkeyVae-v0-level-0_3/DonkeyVae-v0-level-0.pkl

How do I solve this? I train the agent for 5-6 minutes, but no `.pkl` file shows up.

araffin, 2020-01-20:
Hello,

Please fill the issue template completely. I suspect you have a more recent stable-baselines version, the save format is now zip.
I would appreciate a PR that solves that issue ;)

ChunJyeBehBeh, 2020-01-21:
@iamycee You can set cloudpickle=True to save the `.pkl` format file instead of zip format file.

https://stable-baselines.readthedocs.io/en/master/modules/ddpg.html#stable_baselines.ddpg.DDPG.save

iamycee, 2020-01-27:
Thanks @ChunJyeBehBeh 
That solves it. 

Getting another error on `enjoy.py`:

`yaml.constructor.ConstructorError: could not determine a constructor for the tag 'tag:yaml.org,2002:python/object/apply:collections.OrderedDict'
  in "logs/ddpg/DonkeyVae-v0-level-0_4/DonkeyVae-v0-level-0/config.yml", line 1, column 1`

araffin, 2020-01-27:
Yes, this is due to a newer pyyaml version, you need to replace the `yaml.load` by `yaml.load(f, Loader=yaml.UnsafeLoader)` cf rl zoo (https://github.com/araffin/rl-baselines-zoo/commit/028b1cff225c7da679bacac6f2888251bc383be4)

Note that this repo can be seen as a subset of the rl-zoo (which is more up to date).
Again, I would appreciate a PR to fix that.

commit cff4d2ea (2020-02-23): Fixes #21
closed 2020-02-23 by araffin