araffin/learning-to-drive-in-5-minutes #22: Generate the same road

ChunJyeBehBeh, 2020-01-21:
Hello, I would like to train different RL algorithms (DDPG, PPO and SAC) using the same track in the simulator. I tried to go through the issues (#7 and https://github.com/tawnkramer/sdsandbox/issues/15) but still cannot generate the same track every time I start the simulator. 

I even tried to hard code the seed in the code. But still cannot manage to solve it. May I know how to achieve it?
```
def seed(self, seed=None):
        self.np_random, seed = seeding.np_random(10)  #hardcode the seed to 10
        return [seed]
```

araffin, 2020-01-21:
>mean I need to checkout to that branch in order to achieve that?

yes, you do have to use that branch, you also need a newest build of the simulator.

closed 2020-02-03 by ChunJyeBehBeh

KarlXing, 2020-08-07:
Hi @ChunJyeBehBeh @araffin , I also have the need to use the same track. Could you please give some insights for where should I look into? Thanks.

ChunJyeBehBeh, 2020-08-08:
@KarlXing  Do checkout the pretrain branch and there is one parameter 'seed' for you to set. Besides, you need to build latest simulator yourself. Let me know if you face any other issues.