araffin/learning-to-drive-in-5-minutes #26: Training models without VAE [ques]

vijpandaturtle, 2020-04-02:
If you have any questions, feel free to create an issue with the tag [question].  
If you wish to suggest an enhancement or feature request, add the tag [feature request].  
If you are submitting a bug report, please fill in the following details.  

**Describe the bug**
A clear and concise description of what the bug is.

**Code example**
Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

**System Info**
Describe the characteristic of your environment:
 * Commit of your version of the repo
 * GPU models and configuration
 * Python version
 * Tensorflow version
 * Versions of any other relevant libraries

**Additional context**
Add any other context about the problem here.

araffin, 2020-04-16:
Duplicate of #25 

It is possible but not recommended.

closed 2020-04-16 by araffin

Adnan-annan, 2020-12-30:
@araffin training without VAE is not recommended, can you specifically mention because of what reasons ? because it will take more time to train ?

araffin, 2021-01-05:
>because it will take more time to train ?

yes, especially when using SAC, both in term of number of timesteps and wall time.
There are different links in the article that explain why decoupling feature extraction from policy learning may be a good thing ;)