araffin/learning-to-drive-in-5-minutes #27: [Question] How can I get checkpoint while training?

17011813, 2020-05-27:
Hello, I tried to make checkpoint to save the model. 
So I tried to open session with import tensorflow and saver = tf.train.Saver().
But I've got an error 'ValueError' : No variables to save.
I think this is because in train.py code there is no tensorflow session, but I made session on purpose to save checkpoint. 
Also I wanna make pb file too. In pretrained agent that you upload, there is pkl file and parameters, but I want to get checkpoint file and pb file. 
This is the code that I tried. I run **python train.py --algo sac -n 20000 -vae logs/vae-level-0-dim-32.pkl**
with this code, But I can't get checkpoint file,,,
**How can I get checkpoint file with your code?**

`# Code adapted from https://github.com/araffin/rl-baselines-zoo
# Author: Antonin Raffin
import argparse
import os
import time
import warnings
from collections import OrderedDict
from pprint import pprint
​
import tensorflow as tf
from tensorflow.core.protobuf import saver_pb2
​
sess = tf.Session()
LOGDIR = './save'
save_file='./save/model.ckpt'
saver = tf.train.Saver()
​
if os.path.isfile(save_file+".meta"):
   saver=tf.train.Saver()
   saver.restore(sess,save_file)
   print("-----------------run with saved model----------------------------------")
​
else:
   sess.run(tf.global_variables_initializer())
   print("------------------initialize network------------------------------------")
​
# Remove warnings
warnings.filterwarnings("ignore", category=FutureWarning, module='tensorflow')
warnings.filterwarnings("ignore", category=UserWarning, module='gym')
​
import numpy as np
import yaml
from stable_baselines.common import set_global_seeds
from stable_baselines.common.vec_env import VecFrameStack, VecNormalize, DummyVecEnv
from stable_baselines.ddpg import AdaptiveParamNoiseSpec, NormalActionNoise, OrnsteinUhlenbeckActionNoise
#from stable_baselines.ppo2.ppo2 import constfn
​
from config import MIN_THROTTLE, MAX_THROTTLE, FRAME_SKIP,\
   MAX_CTE_ERROR, SIM_PARAMS, N_COMMAND_HISTORY, Z_SIZE, BASE_ENV, ENV_ID, MAX_STEERING_DIFF
from utils.utils import make_env, ALGOS, linear_schedule, get_latest_run_id, load_vae, create_callback
from teleop.teleop_client import TeleopEnv
​
parser = argparse.ArgumentParser()
parser.add_argument('-tb', '--tensorboard-log', help='Tensorboard log dir', default='', type=str)
parser.add_argument('-i', '--trained-agent', help='Path to a pretrained agent to continue training',
                   default='', type=str)
parser.add_argument('--algo', help='RL Algorithm', default='sac',
                   type=str, required=False, choices=list(ALGOS.keys()))
parser.add_argument('-n', '--n-timesteps', help='Overwrite the number of timesteps', default=-1,
                   type=int)
parser.add_argument('--log-interval', help='Override log interval (default: -1, no change)', default=-1,
                   type=int)
parser.add_argument('-f', '--log-folder', help='Log folder', type=str, default='logs')
parser.add_argument('-vae', '--vae-path', help='Path to saved VAE', type=str, default='')
parser.add_argument('--save-vae', action='store_true', default=False,
                   help='Save VAE')
parser.add_argument('--seed', help='Random generator seed', type=int, default=0)
parser.add_argument('--random-features', action='store_true', default=False,
                   help='Use random features')
parser.add_argument('--teleop', action='store_true', default=False,
                   help='Use teleoperation for training')
[... 181 lines cut ...]
# Save hyperparams
with open(os.path.join(params_path, 'config.yml'), 'w') as f:
   yaml.dump(saved_hyperparams, f)
​
if args.save_vae and vae is not None:
   print("Saving VAE")
   vae.save(os.path.join(params_path, 'vae'))
​
if normalize:
   # Unwrap
   if isinstance(env, VecFrameStack):
       env = env.venv
   # Important: save the running average, for testing the agent we need that normalization
   env.save_running_average(params_path)
​
​
if not os.path.exists(LOGDIR):
   os.makedirs(LOGDIR)
checkpoint_path = os.path.join(LOGDIR, "model.ckpt")
filename = saver.save(sess, checkpoint_path)
print("Model saved in file: %s" % filename)
​
tf.io.write_graph(sess.graph_def, '.', 'graph.pb', as_text=False)​`

I'm really appreciate with your this project!!! Thanks a lot 👍

araffin, 2020-05-27:
Hello,
The answer is the documentation of stable-baselines (cf saving/loading in stable-baselines.readthedocs.io/), you need to use a callback to make checkpoints.

We have already a callback saving the best model: https://github.com/araffin/learning-to-drive-in-5-minutes/blob/master/train.py#L195

Note that the code it out of data and does not match the latest SB version. I plan to open source an updated version in the coming month but I won't update this repo.

PS: please use markdown to format your code next time

17011813, 2020-05-28:
Thanks so much : )