araffin/learning-to-drive-in-5-minutes #28: stable-baselines version requirement is incorrect

mechovation, 2020-05-27:
requirements.txt lists:

stable-baselines>=2.9.0

pip install of stable-baselines resulted in 2.10.0, meeting the listed requirements, but resulting in this error:

Traceback (most recent call last):
  File "train.py", line 19, in <module>
    from stable_baselines.ppo2.ppo2 import constfn
ImportError: cannot import name 'constfn'

recommend updating requirements.txt to reflect stable-baselines==2.9.0

araffin, 2020-05-27:
Hello,
Thanks for pointing that out.
I would appreciate a pull request that either:
1. Fix the requirements to 2.9.0
2. Do the changes to make it compatible with 2.10.0 (**preferred**) and set the minimum version to 2.10.0

Jekyll555, 2020-06-09:
I fixed it like this (taking some change folder infos from stable baselines docs)
```
-from stable_baselines.ppo2.ppo2 import constfn
+from stable_baselines.common.schedules import constfn
```

araffin, 2020-06-09:
As mentioned before, I would appreciate a PR that solves that issue ;)
(there will be also some changes in SAC that needs to be done)

PR #32 (2020-11-06): Fix for new Stable-Baselines version
updates constfn import in train.py, total_episode_reward_logger import in custom_ppo and custom_sac, and sets min SB version to 2.10.0

closes #28
commit ccb27e66 (2020-11-06): Fix for new Stable-Baselines version (#32)

* Fixes #28

updates constfn import in train.py, total_episode_reward_logger import in custom_ppo and custom_sac, and sets min SB version to 2.10.0

* more Stable Baselines refactoring changes

* run() to _run() in line with stable baselines v2.10.0
closed 2020-11-06 by araffin