araffin/learning-to-drive-in-5-minutes #29: Not connecting to simulator [bug]

josiahcoad, 2020-06-04:
Hello, I am trying to run your code exactly as is (using your pretrained agent and vae) and just get :__"Waiting for sim to start...if the simulation is running, press EXIT to go back to menu.__". I also tried setting the  export DONKEY_SIM_PATH environment variable but to no avail.

**Code example**
>>>  python enjoy.py --algo sac -vae vae-level-0-dim-32.pkl --exp-id 6 -n 2000

**System Info**
Describe the characteristic of your environment:
 * Commit of your version of the repo: 89a3b2ca040014cb2193ad3fe88636de146f49ce
 * GPU models and configuration:  GeForce GTX 1070
 * Python version: 3.7
 * Tensorflow version: 1.13.1

araffin, 2020-06-05:
Hello,
There is one important info that is missing: what simulator are you using?
And did you launch it before launching the script?

josiahcoad, 2020-06-05:
Using donkey sim "SDSandbox v 20.5.16". I've did try launching sim before running script

araffin, 2020-06-05:
>Using donkey sim "SDSandbox v 20.5.16". I've tried several things:

This is a newer version compared to the one tested. Did you try using the one from the README?

josiahcoad, 2020-06-05:
Oh I see, that's unfortunate. the linked sim in the README is just for linux (unless I were to build it myself). Looking forward to your next release!

araffin, 2020-06-05:
You can also try older binaries from the release page: https://github.com/tawnkramer/gym-donkeycar/releases

September 2018 may work: https://github.com/tawnkramer/gym-donkeycar/releases/tag/v18.9

There are binaries for Linux, Mac and Windows.

josiahcoad, 2020-06-05:
Well, I did get a different result. Now I get (after setting the env var DONKEY_SIM_PATH)
```
Got a new client ('127.0.0.1', 58585)
Scene Selection Ready
Connection dropped
```