araffin/learning-to-drive-in-5-minutes #30: Model taking 10 minutes to load? [question]

josiahcoad, 2020-06-04:
Hello, I am trying to run your code exactly as is (using your pretrained agent and vae) and the model takes ~10 minutes to load. I was just wondering if this was normal.

**Code example**
>>>  python enjoy.py --algo sac -vae vae-level-0-dim-32.pkl --exp-id 6 -n 2000

**System Info**
Describe the characteristic of your environment:
 * Commit of your version of the repo: 89a3b2ca040014cb2193ad3fe88636de146f49ce
 * GPU models and configuration:  GeForce GTX 1070
 * Python version: 3.7
 * Tensorflow version: 1.13.1

**Additional context**
Add any other context about the problem here.

araffin, 2020-06-05:
Hello,

No, this is not normal, it should take only some seconds (once tensorflow is loaded).
 as in #29 what simulator are you using?
and what is your OS?

josiahcoad, 2020-06-05:
Thank you for your response! So on my MacOS the model loads very quickly... just Windows 10 it has issues (tried it on two different Windows 10 computers). I mentioned sim in #29

araffin, 2020-06-05:
>.. just Windows 10 it has issues (tried it on two different Windows 10 computers)

This was only tested on linux. I cannot give any guarantees for windows.