araffin/learning-to-drive-in-5-minutes #31: Collecting Images For VAE [question]

KarlXing, 2020-08-07:
Hi, thank you for this interesting work. I'd like to collect images for the VAE training instead of using the pretrained VAE model. May I know how many samples did you collect for each level? Are there any suggestions about the manual control during data collection?  I appreciate it for any advice on that. Thanks :)

araffin, 2020-08-07:
Hello,
The number of images does not really matter, what matter is that you cover most of the observation distribution, so it's good to do not only expert driving ;)
Best is to check visually the trained autoencoder and to wait until convergence.

If you want number, I think I had something working with ~5k images or less (so ~5-10 minutes driving).

KarlXing, 2020-08-07:
That makes sense. Thanks!

closed 2020-08-07 by KarlXing