araffin/learning-to-drive-in-5-minutes #33: Training for collision avoidance of small car

Adnan-annan, 2021-02-02:
Hi, i am trying to use Jetbot for collision avoidance using a small car with your approach. I place a small car at the center of the track and try to train it to avoid the car using your approach. The results are very unstable, sometimes it seems to work but again after few more episodes, it forgets the learning. Do you have any suggestions for this task ? Do you recommend any changes to the approach? Other thing is that it takes long time to train. Kindly provide me some recommendations to try. I shall be very thankful to you.

araffin, 2021-02-02:
> The results are very unstable, sometimes it seems to work but again after few more episodes, it forgets the learning. Do you have any suggestions for this task ?

you probably need to give a more informative reward + augment the observation if possible.
But before, you need to make sure that the autoencoder is sufficiently trained (so that the car can see the other one to avoid it).

An example of collision avoidance system (using a 1D lidar): https://stable-baselines.readthedocs.io/en/master/misc/projects.html#train-a-ros-integrated-mobile-robot-differential-drive-to-avoid-dynamic-objects

Adnan-annan, 2021-02-03:
Could you please elaborate a little bit about what is informative reward,,, also by augmenting the observation do you mean augmenting the input of VAE model before training it ? and what kind of augmentations do you recommend for this task?  Currently i am just exploring to use your strategy for collision avoidance of small car on a track using just the images (VAE output) as input (not using the LIDAR) for a Jetbot. I shall be very grateful if you could suggest some modification or adjustments to your strategy for this take to make it work robustly. I am using  https://github.com/masato-ka/airc-rl-agent for training on SAC RL algorithm on jetbot which is based entirely on your method.

Adnan-annan, 2021-02-05:
@araffin any further pointers and suggestions will be very helpful