araffin/learning-to-drive-in-5-minutes #34: self.obs doesn't change

eliork, 2021-02-02:
https://github.com/araffin/learning-to-drive-in-5-minutes/blob/ccb27e66d593d6036fc1076dcec80f74a3f5e239/algos/custom_ppo2.py#L157

Hi, I am trying to recreate your approach in a custom environment I have built. I am trying to use your custom version of PPO2, but I realize that after [this line](https://github.com/araffin/learning-to-drive-in-5-minutes/blob/ccb27e66d593d6036fc1076dcec80f74a3f5e239/algos/custom_ppo2.py#L157) self.observation doesn't change. is that an expected behaviour?

Thanks, appreciate your work

araffin, 2021-02-03:
>realize that after this line self.observation doesn't change. is that an expected behaviour?

why should it change?
with VecEnv, the reset is automatic (cf doc).

eliork, 2021-02-03:
shouldn't ```self.obs``` hold the current observation? isn't that the expected output from the ```self.env.step(clipped_actions)``` function?

araffin, 2021-02-03:
>shouldn't self.obs hold the current observation?

`self.obs` holds the new observation after stepping in the env.

eliork, 2021-02-09:
I found the problem, I guess I configured the observation space wrong. I configured it as 

```
self.observation_space = spaces.Box(low=np.finfo(np.float32).min,
                              high=np.finfo(np.float32).max,
                              shape=(1, self.z_size + self.n_commands * self.n_command_history),
                              **dtype=np.uint8)**
```
instead of having ```dtype=np.float32```
Thank you for your help, closing this issue

closed 2021-02-09 by eliork